WorldofKerry/Python2Verilog #4: Use `localparam` in Verilog Instead of Variables for Constants

WorldofKerry, 2023-07-11:
Instead of using variables for constants, use `localparam` in Verilog.

henrydavies1, 2023-07-16:
Can I pick this one up?

WorldofKerry, 2023-07-16:
Yea for sure! I've updated the issue as I've had some other thoughts since its creation.

henrydavies1, 2023-07-17:
Thanks for the updated info, I've got a few questions:

- I'm a bit of a Python noob, but after some googling it seems that you can't have 'proper' constants. You can name things in capital snake case but it's just a convention. Should I infer a variable is a constant if it follows the convention?
- From looking at the code the only way to add variables is with `add_global_var`. Is there any other way?
- Why would a  `Context` class be needed? I was thinking we'd just need to extend `parse_statement` to output `localparam` for constants.

WorldofKerry, 2023-07-18:
> * I'm a bit of a Python noob, but after some googling it seems that you can't have 'proper' constants. You can name things in capital snake case but it's just a convention. Should I infer a variable is a constant if it follows the convention?
Yes, CAPITAL_CASE can be inferred to be constant (although I'm not too good at remembering to make my constants cap case). I believe Pylint will catch instances where a CAPTICAL_CASE var is modified. 

> * From looking at the code the only way to add variables is with `add_global_var`. Is there any other way?
Currently that is the only way. 

> * Why would a  `Context` class be needed? I was thinking we'd just need to extend `parse_statement` to output `localparam` for constants.
Right now [this line](https://github.com/WorldofKerry/Python2Verilog/blob/fc90d902600f5e3965abfec659ef70618b4db7fa/python2verilog/convert.py#L67C58-L67C58) uses the `.global_vars`. I expect additional properties need to be passed along in the future, including `localparam`s.

henrydavies1, 2023-07-23:
Thanks for your response, that all makes sense. I can see you've created a `Context` class so I can utilise that.

I want to add a test to `tests/frontend/test_generator_parser.py`, but none of the tests in there seem to test anything at the moment?

WorldofKerry, 2023-07-23:
> I want to add a test to `tests/frontend/test_generator_parser.py`, but none of the tests in there seem to test anything at the moment?
You're right! Currently, that's just a place for me to print the IR for manual inspection. I've found it quite difficult to have more automated testing there, as assertions on how the IR is structured do not directly relate to getting the correct outputs with the Verilog simulation. In `tests/integration/`, you'll see more robust tests that will actually fail from errors (i.e. fail if Verilog output does not match Python output).

I'm currently restructuring GeneratorParser to have a Graph representation, as I've run into some architectural problems with the optimization algorithms. Perhaps it may be more ideal if you began with `backend/verilog.py`, and implementing some sort of "local param" as a part of the `Module` class. In the end, I think it should be possible to generate the wrapper around the body, with only the Context being passed to the Verilog generator.

henrydavies1, 2023-07-31:
Thanks that makes sense. Hopefully I'll be able to pick this up this week.

henrydavies1, 2023-08-06:
Hi, sorry I haven't made much progress with this. After looking at the code and doing some more learning of Verilog I've realised the code changes I'll have to make are quite a lot more than I expected when I first looked at the ticket. I'm going to leave this for someone else to pick up, sorry for the hassle and thanks for answering my questions.

WorldofKerry, 2023-08-10:
#67 Resolves

closed 2023-08-10 by WorldofKerry