WorldofKerry/Python2Verilog #134: Append _H or _L for active high/low signals

Haxrox, 2023-09-12:
See https://github.com/WorldofKerry/Python2Verilog/pull/119#discussion_r1323252933 as well

_Originally posted by @Haxrox in https://github.com/WorldofKerry/Python2Verilog/pull/119#discussion_r1320932691_

The idea is by appending _H or _L to your ready, valid, etc signals, this allows users to use them as argument names in Python. This should allow the removal of `_` before each parameter when defining the module in Verilog, allowing users to easily use `.*` to instantiate modules. 

However, this can lead to the issue where users use _H or _L convention when defining argument names in Python, in which case you can either
a) throw an error and tell the user to pick another name
b) mangle by appending _

WorldofKerry, 2023-09-12:
I think it would be better if the user can config the default signal names.

Then throw an error and tell the user to pick another name if they fail to config without reusing names.

WorldofKerry, 2023-09-14:
Duplicate of #9

closed 2023-09-14 by WorldofKerry